SHA2017-badge/Firmware #8: Question: New distribution technique

slaff, 2017-05-21:
In [this post](https://sha2017.org/blog/talk-announcement-the-sha2017-badge-the-badge-is-a-go) you are mentioning 

> Featuring a brand new distribution technique designed by Sprite_TM, we can update badges without associating to the WiFi access points.

Can describe with technical details what is meant here?

annejan, 2017-05-23:
This is the bpp update protocol suite hackery thingamajig. It's essentially a protocol encapsulating
multiple streams, sent over UDP over a WiFi network, so clients can pick it up and decode it without
having to connect to the network or transmit anything.

https://github.com/SHA2017-badge/bpp

closed 2017-05-27 by niekproductions

slaff, 2017-10-10:
@annejan Thanks a lot!