SHA2017-badge/Firmware #13: Intermittent build failures: no rule to make target .../bootloader.bin

raboof, 2017-05-27:
e.g. https://travis-ci.org/SHA2017-badge/Firmware/builds/236725156

    make[1]: *** No rule to make target `/home/travis/build/SHA2017badge/Firmware/build/bootloader/bootloader.bin'.  Stop.

annejan, 2017-05-30:
and again https://travis-ci.org/SHA2017-badge/Firmware/builds/237452747

Hadn't seen this happening before . . will try and find cause, might have something to do with "ugly" hack in Makefile to source paths . .

raboof, 2017-05-30:
Yeah it's weird, esp. since it seems 'consistent' in the sense that restarting a build yields the same result each time, but committing something that shouldn't have any impact may make it succeed...

annejan, 2017-05-30:
Testing on a clean Ubuntu VM

Want to be able to reproduce this . . perhaps we should remove the hack from Makefile and add a `source set_env.sh` step?

raboof, 2017-05-30:
Worth a try I suppose, though I'm not sure I see how it could be related...

How about adding a 'make clean' to `.travis.yml`?

annejan, 2017-05-30:
Unfortunately this is not a very good test . . 
https://github.com/SHA2017-badge/Firmware/commit/cdbd74d861d889cf4e74c2a5d067207a341b9968

https://travis-ci.org/SHA2017-badge/Firmware/builds/237460784

commit 4c179a70 (2017-05-30): Merge pull request #16 from SHA2017-badge/make-magic-cleanup

Attempt to stabalize build (fixes #13)
closed 2017-05-30 by annejan