SHA2017-badge/Firmware #15: UGFX 'toggle' input driver

raboof, 2017-05-27:


annejan, 2017-05-27:
Can you explain what you mean with this issue.
It's not very descriptive . .

raboof, 2017-05-27:
Yeah, sorry, just quickly jotted down what I was planning to do :)

uGFX has a subsystem for input, GINPUT, for sending button presses to widgets on windows. To use that we'd need to wire up the low-level button detection (already functional in `main/main.c`) with uGFX though a sort of 'driver'.

annejan, 2017-06-09:
It seems using the GINPUT features needs all the Widget / WM stuff too . .

I don't know if we want to go that route or have a simple `badge.getinput(<timeout>)` function instead . .

Some work on getting GINPUT working can be found at https://github.com/SHA2017-badge/Firmware/tree/aj-input-test

annejan, 2017-06-10:
Merged https://github.com/SHA2017-badge/Firmware/pull/22 needs some more love <3

raboof, 2017-06-10:
Current status:
* low level driver added gets triggered on input interrupt and updates the button state
* tested from `demo_ugfx.c` with ancient badge (check the logs, shows the state of the 'up' button each time the display is refreshed)

Next up:
* test with a new badge (but I don't have one myself currently)
* test if this can be exposed as a `GEVENT` 'source' (https://wiki.ugfx.io/index.php/GEVENT_FAQ#What_is_a_source.3F) and hooked up to a listener
* expose to upy

basvs, 2017-06-11:
Done:

- refactor to do less lowlevel gpio/pin stuff in ginput_lld_toggle.c, do that in badge/ instead

- test with a new badge (but I don't have one myself currently)

raboof, 2017-06-11:
exposing as GEVENT works, demo'd in #26. Not really familiar with they python stuff yet myself, not currently planning to pick that up so whoever is interested feel free to give it a shot :)

neuhaus, 2017-06-12:
I'm getting a build error during `make`, is it related to this issue?

Here's the error:

```
CC gfx_mk.c
In file included from ../../ugfx/src/ginput/ginput_mk.c:10:0,
                 from gfx_mk.c:16:
../../ugfx/src/ginput/ginput_keyboard.c: In function ‘_gkeyboardInitDriver’:
../../ugfx/src/ginput/ginput_keyboard.c:504:5: error: this ‘if’ clause does not guard... [-Werror=misleading-indentation]
     if (!gkvmt(k)->init((GKeyboard *)g, driverinstance))
     ^~
../../ugfx/src/ginput/ginput_keyboard.c:508:2: note: ...this statement, but the latter is misleadingly indented as if it is guarded by the ‘if’
  if (!gtimerIsActive(&KeyboardTimer))
  ^~
cc1: all warnings being treated as errors
../py/mkrules.mk:47: die Regel für Ziel „build/gfx_mk.o“ scheiterte
make: *** [build/gfx_mk.o] Fehler 1

```

annejan, 2017-06-12:
It's semi-related and something I ran into too, I'm thinking it's a warning that only happens on recent versions of GCC, will try and fix it this evening . .

annejan, 2017-06-12:
This should be fixed @neuhaus, when you pull Firmware and run a `submodule update` in the Firmware folder. It was an error in the upstream `ugfx` submodule.

annejan, 2017-06-12:
I'm closing this issue, when issue's arise just open a new one 👍

closed 2017-06-12 by annejan

neuhaus, 2017-06-12:
Thanks for the quick help!