SHA2017-badge/Firmware #51: Support for wake-up events or keeping a background process running periodically

lkarlslund, 2017-06-21:
It would be real nice with a way to disable sleep for the badge with a call from python, or somehow to ask the badge to wake up and trigger an event after a set time.

Something like:

badge.disable_sleep()
badge.wakeup_after(xxx) where xxx is a millisecond delay

annejan, 2017-07-01:
There are two ways to do this we are considdering.

A) writing current program and possibly other state info to the RTC memory.

B) using a json file to configure what should be ran on boot.

I'm personally a bigger fan of using A so doing a hard reset won't get us into stupid boot loops if the json points to something broken.

annejan, 2017-07-08:
This now works as follows.
bytes of rtcmem
0 = what to do
1 = inverse of 0
2->\0 is a string

Currently on mismatch of 0 and 1 micropython boots and imports the string!
1 and 254 start OTA update process

closed 2017-07-08 by annejan