SHA2017-badge/Firmware #59: Make sure the WS2812_DATA is in high-impedance mode before the LED power is switched off

annejan, 2017-06-26:
It might break 📟

basvs, 2017-06-27:
on disabling the power to the ws2812/sk6812, i can disable the spi and configure the gpio pin as input without pull-up or pull-down. i think that qualifies as high-impedance.

should i do this right before or right after disabling power?

niekproductions, 2017-06-27:
Probably before disabling power, otherwise the sk6812's input capacitance can drain into the ESP and damage its I/O

basvs, 2017-06-28:
Oh, and can someone with SK6812RGBW leds test the leds demo? It should start with these 6 leds: red, green, blue, white, black, black..

annejan, 2017-06-28:
I'll test when I get home..

annejan, 2017-06-28:
And . . it's Green, Red, Blue, Black, Black

basvs, 2017-06-28:
Added pull-request #62

basvs, 2017-06-28:
sprite_tm: is this the best way to put a GPIO pin in high-impedance?

Spritetm, 2017-06-29:
There isn't really a 'the best way'; this one should do just fine.

closed 2017-07-03 by basvs