SHA2017-badge/Firmware #70: Freeing the HSPI bus breaks the VSPI config.

basvs, 2017-07-03:
I've currently commented out this code in badge_leds_disable() to work around the problem. Definitely not the best solution, but now the code still seems to work. (I **assume** that the port is successfuly configured as GPIO INPUT port, but I am not sure..)

//  FIXME: freeing the HSPI seems to (de)configure the VSPI as well..
//  ret = spi_bus_free(HSPI_HOST);
//  assert( ret == ESP_OK );

(I have mention the issue in issue #59 and/or pull-request #62)

basvs, 2017-07-31:
When I disable these lines in components/driver/periph_ctrl.c, the VSPI keeps working:

    case PERIPH_HSPI_MODULE:
        DPORT_CLEAR_PERI_REG_MASK(DPORT_PERIP_CLK_EN_REG, DPORT_SPI_CLK_EN);
        DPORT_SET_PERI_REG_MASK(DPORT_PERIP_RST_EN_REG, DPORT_SPI_RST);

Spritetm, 2017-07-31:
Fwiw, someone did an oopsie in the register descriptions for the peripheral mapping. We have a fix ready, it's supposed to hit Github as soon as CI is done.

basvs, 2017-07-31:
I've added pull-request #169 to work around it and implement my hack.

Waiting for upstream to make a nice fix. :)

Spritetm, 2017-07-31:
https://github.com/espressif/esp-idf/commit/9314bf0d376fc3dfdc5b954376a40f74fd4a1d78 <- upstreams 'nice' fix :)

basvs, 2017-07-31:
Upstream fix tested and works! still needs the changes in components/badge/badge_leds.c from pull-request #169 . The hack in https://github.com/SHA2017-badge/esp-idf/pull/5 is no longer needed.

annejan, 2017-07-31:
This should be fixed now, can you confirm, @basvs ?

basvs, 2017-08-01:
All code is in master, so should be fixed.

closed 2017-08-01 by basvs