SHA2017-badge/Firmware #73: Skip frames when flushing

raboof, 2017-07-04:
When flushing 3 times immediately after each other, the first flush will be instant, while the second 2 flushes will both take about 33 ticks.

Apparently flushing is somehow asynchronous. It would be neat if we could somehow detect the first flush is still in progress, and skip the second flush when the third flush comes in before the first flush has really finished.

annejan, 2017-07-04:
We could probably expose the "bussy" pin state to the micropython env.

Something like badge.eink_bussy()   True False

raboof, 2017-07-04:
we can even do that entirely on the C (ugfx lld) side I'd say - but yeah, worth a try.

annejan, 2017-07-04:
It used to be fully synchronous,  which wasted a lot of time after sending a flush()

I don't think ugfx really has something to manage the wait in a satisfactory way, that's why I guess doing the checking high-level would be most ideal.

annejan, 2017-07-04:
This should work https://github.com/SHA2017-badge/micropython-esp32/pull/36

commit d64faf00 (2017-07-04): Export badge.eink_busy() and badge.eink_busy_wait() closes #73
closed 2017-07-04 by annejan