SHA2017-badge/Firmware #83: Timeout on sdcard after sdcard-power-cycle

basvs, 2017-07-08:
Current order:
- enable power to sdcard
- mount sdcard with esp_vfs_fat_sdmmc_mount()
- read/write files
- unmount sdcard with esp_vfs_fat_sdmmc_unmount()
- disable power to sdcard

When doing a 2nd mount (power-on, mount, unmount, power-off, power-on, mount), the initialization fails with a timeout. It sounds like I have to initialize something extra.

basvs, 2017-07-08:
Selected 'demo sd-card image'
badge_power: enabling power to sdcard and leds.
D (199838) intr_alloc: Connected src 37 to int 17 (cpu 0)
Name: SD16G
Type: SDHC/SDXC
Speed: default speed
Size: 14804MB
CSD: ver=1, sector_size=512, capacity=30318592 read_bl_len=9
SCR: sd_spec=2, bus_width=5
badge_power: disabling power to sdcard and leds.

Selected 'demo sd-card image'
badge_power: enabling power to sdcard and leds.
D (214528) intr_alloc: Connected src 37 to int 17 (cpu 0)
E (214698) sdmmc_cmd: sdmmc_card_init: send_scr returned 0x107
E (214698) example: Failed to initialize the card (263). Make sure SD card lines have pull-up resistors in place.

basvs, 2017-07-09:
More testing:
- re-insert sdcard → sdcard works
- restarting 'make monitor' (doing a soft reboot) → sdcard works

annejan, 2017-07-15:
Can not reproduce . .

basvs, 2017-07-15:
Still reproducible here. I guess your sdcard brand is better. :-)
But I created a fix. Patch is on its way.

basvs, 2017-07-15:
fixed with pull request #97

closed 2017-07-15 by basvs