SHA2017-badge/Firmware #99: 'make' command ends with 2 errors

archonoff, 2017-07-16:
I'm trying to install badge emulator on OS X 10.11.6 using instruction from https://wiki.sha2017.org/w/Projects:Badge/Emulator#macOS and getting these errors:

```
../../ugfx/src/gos/gos_linux.c:108:16: error: use of undeclared identifier 'CLOCK_MONOTONIC'
        clock_gettime(CLOCK_MONOTONIC, &ts);
                      ^
../../ugfx/src/gos/gos_linux.c:211:20: error: use of undeclared identifier 'CLOCK_REALTIME'
                                        clock_gettime(CLOCK_REALTIME, &tm);
```

annejan, 2017-07-16:
Interesting, that must be some missing posix library that is installed on my Mac . .
I'll see if I can reproduce this issue and remedy it . .

archonoff, 2017-07-16:
@annejan 
Do you have macos 10.12? It looks like this problem occurs only for systems' versions lower than 10.12.
But I can't install Sierra on my laptop: appstore says it's too old.
Maybe is it possible to use some solutions from [this thread](https://stackoverflow.com/questions/5167269/clock-gettime-alternative-in-mac-os-x)?

annejan, 2017-07-17:
Yes, running macos 10.12.6 beta myself

Please use debian / ubuntu VM if toolchain not working . .

closed 2017-07-17 by annejan

basvs, 2017-07-18:
https://stackoverflow.com/questions/5167269/clock-gettime-alternative-in-mac-os-x

this might help.

annejan, 2017-07-18:
I would advice upgrading from outdated macOS to Debian 😈  or FreeBSD . .

Ch3pa, 2017-07-18:
Sadly my laptop on debian is out of order. 
I tried to run the emulator on docker/jessie though but I struggle with SDL2. 
Perhaps a VM is more wise.

annejan, 2017-07-18:
After this weekend we should have a V0 master firmware.

That might also be the moment to release the emulator in binary form..

fbettag, 2017-08-06:
Edit micropython/unix/Makefile and set OPT_OS from "linux" to "osx". Not sure why this is hardcoded.

fbettag, 2017-08-06:
Also in ugfx/boards/base/Linux-SDL/board.mk someone hardcoded

```
GFXDEFS += -DGFX_USE_OS_LINUX=TRUE
```
which then produces

```
../../ugfx/src/gos/gos_rules.h:24:3: error: "GOS: More than one operation system has been defined as TRUE."
```

fbettag, 2017-08-06:
Also one needs to define

```
#define gfxTicksToMilliseconds(ticks)	(ticks)
```
in ugfx/src/gos/gos_osx.h in order to get the unknown symbol in there. Not sure tho if we need to multiply it on OSX like on freertos.