SHA2017-badge/Firmware #140: emulator build fails with current instructions

eternallyBaffled, 2017-07-25:
Simply following the instructions on the wiki in a fresh environment yields:
```
root@Junix-laptop:~/sha2017/Firmware/micropython/unix# make
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
mkdir -p build/genhdr
Generating build/genhdr/mpversion.h
GEN build/genhdr/qstr.i.last
In file included from ../py/../extmod/modussl_mbedtls.c:36:0:
./esp_log.h:8:23: fatal error: sdkconfig.h: No such file or directory
#include "sdkconfig.h"
                      ^
compilation terminated.
In file included from esp_log.c:5:0:
esp_log.h:8:23: fatal error: sdkconfig.h: No such file or directory
#include "sdkconfig.h"
                      ^
compilation terminated.
../py/mkrules.mk:73: recipe for target 'build/genhdr/qstr.i.last' failed
make: *** [build/genhdr/qstr.i.last] Error 1
make: *** Deleting file 'build/genhdr/qstr.i.last'
```

basvs, 2017-07-25:
We have to fix the documentation or supply a known good sdkconfig.h. As far as I know, the sdkconfig.h is currently only used for the loglevel. (I'm afraid I am the one responsible for this change..)

Easiest way to fix this is to build a firmware binary first:
    cd Firmware
    make defconfig
    make
    cd micropython/unix
    make

annejan, 2017-07-25:
The fix can just be `#ifndef UNIX` around the `#include`

basvs, 2017-07-25:
@annejan : that doesn't work anymore. I also placed a lot of ESP_LOGx() entries in the ussl_mbed code

basvs, 2017-07-25:
This might also work: (haven't tested this)

    cd Firmware/micropython/unix
    touch sdkconfig.h
    make

annejan, 2017-07-25:
That should just break on the unix build too . . (spoiler they were already `#ifdef-ed` out of the emu build)

basvs, 2017-07-25:
This fixed the ESP_LOGx() problems in the zlib inflater:

https://github.com/SHA2017-badge/micropython-esp32/commit/2b5d93822a68eeb06c107d64a0c4249373402806

I later added this patch to add more warnings to the tls implementation:

https://github.com/SHA2017-badge/micropython-esp32/commit/09be7ffc6a60e47ea8f9a0548017fda15e3a6ec5

annejan, 2017-07-26:
This has been resolved yesterday, sorry for not getting back to you about the fix @eternallyBaffled

closed 2017-07-26 by annejan