SHA2017-badge/Firmware #183: Changing the display

emilekm2142, 2017-08-04:
I want to build one badge on my own, but I would like to use St7735 based display. Ugfx has a driver for it. What should I do to make it work? Should I edit the files in `components\ugfx-glue`?

basvs, 2017-08-07:
We're only using the ugfx as an intermediate layer. I don't know if you can easily replace the ugfx driver.

The badge firmware works well with both the GDEH029A01 and the DEPG029B01 eink screens. Why not use a GDEH029A01 screen? :-)

raboof, 2017-08-07:
Ah that's LCD rather than eink. The badge was optimized for eink, but I don't see why you shouldn't be able to make this work. Indeed replacing some of `components\ugfx-glue` sounds like it might be sufficient. Curious to hear how it works out!

emilekm2142, 2017-08-07:
Unfortunetly, I am not proficient enough in C to do that :/. Being able to change the display easily would make this firmware a base for a whole bunch of python-based cool stuff ;)

annejan, 2017-08-21:
I have some LCD modules laying around, might be able to squeeze in a couple of hours to try this on the ESP32 dev kit + LCD breakout(s) somewhere over the next two weeks.

I'm 90% certain I have a ST7735 laying around from experiments with some other SOCs . .

emilekm2142, 2017-08-21:
Thank you annejan! I will praise you till the end of my days if you menage to do that 😍