SHA2017-badge/Firmware #184: building on OSX

nlitsme, 2017-08-05:
The toolchain from the (xtensa-esp32-elf)[https://github.com/SHA2017-badge/xtensa-esp32-elf] repository is for Linux only.

If you want to build on a mac, do the following:

* download the (mac toolchain)[https://dl.espressif.com/dl/xtensa-esp32-elf-osx-1.22.0-59.tar.gz] from (this page)[http://domoticx.com/sdk-esp32-xtensa-architecture-toolchain/]
* rename the linux toolchain: `mv xtensa-esp32-elf xtensa-esp32-elf-linux`
* untar the mac toolchain: ` tar zxf ~/Downloads/xtensa-esp32-elf-osx-1.22.0-59.tar.gz`

Roosted7, 2017-08-07:
Could you make a pr and update the readme? :+1:

basvs, 2017-08-07:
@annejan iirc you use osx as well. can you help out? :)

nlitsme, 2017-08-09:
here is my pull request: https://github.com/SHA2017-badge/Firmware/pull/202

closed 2017-08-16 by annejan