SHA2017-badge/Firmware #193: Prevent sleeping when connected to power

the-louie, 2017-08-07:
It would be nice if the badge stayed awake when connected to an external power source. It is a bit of a hassle trying to develop apps and upload to test when the badge falls asleep.

Roosted7, 2017-08-07:
The badges have a BPP update feature that should allow them to be updated ota while 'sleeping' and not associated with wifi. Atm this is not activated, but if badges do not 'sleep' they will not receive those updates. They do not sleep to reduce power consumption..

If you type 'import shell' into the repl the badge won't start the splash and won't go to sleep at all. You should try it for developing apps :)

closed 2017-08-07 by Roosted7

basvs, 2017-08-07:
@the-louie Since 'Sinistere Site' it is possible to stay awake. It is a debug-feature and not meant for production.

You can enable it with:

    badge.nvs_set_u8('badge', 'usb_stay_awake', 1)

basvs, 2017-08-07:
The reason for not staying awake in production: Some people use a power-bank to power the badge. If the badge won't sleep, this would drain the power-bank.

the-louie, 2017-08-10:
@basvs excellent, will try this as soon as I get back to playing with the badge.