SHA2017-badge/Firmware #201: sha2017_ota.c: strncpy misuse (potential buffer overflow)

pesco, 2017-08-08:
the line

      strncpy(dest, text + i, len);

would copy an HTTP response line of arbitrary length into the fixed-size buffer `dest` if it weren't for the saving fact that the `text` buffer is of the same size as `dest`. the length argument to strncpy should be the size of `dest`.

similarly in line 195:

      strncpy(dest, dest + 16, len - 17);

in addition, the behavior of `strncpy` is undefined for overlapping arguments.

basvs, 2017-08-12:
The ota code needs some rewriting.. I hope to find some time for it..

basvs, 2017-08-12:
The ota code was buggy and completely unreadable; tried to make it bug-free and readable again:

https://github.com/SHA2017-badge/Firmware/pull/205

basvs, 2017-08-16:
@pesco This file has largely been rewritten. Can I close this issue?

pesco, 2017-08-17:
If the code in question no longer exists, sure.

closed 2017-08-17 by basvs