SHA2017-badge/Firmware #207: V_bat is measured / displayed with +0.4V Offset

geek-solutions, 2017-08-15:
The Voltage of the battery is measured and displayed with an offset of +0.4V (values from multimeter: 3.8V badge: 4,2V). Therefore the battery indicator also shows a wrong status for the battery.

Only used one (my) badge for testing

basvs, 2017-08-15:
Every badge seems to be a bit different. The only right fix might be to build some kind of calibration for it..

basvs, 2017-08-15:
@Roosted7 If I remember correctly, you were trying to find out how we could solve this?

renzenicolai, 2018-01-11:
HW bug with bad (software) fixes on top: the analog input used to measure the battery voltage is connected AFTER the fuse. The resistance of the fuse causes a voltage drop which differs from badge to badge and from load to load (with the leds or vibration motor on the voltage drop is higher...).
So... only way to fix is to disconnect the ADC pin and connect it to the battery directly.

Roosted7, 2018-01-11:
What renze says is true, but not the biggest cause of the false readings... That is mostly because the ESP32's ADC is _not so good_ :sweat: (Especially in the old IDF's at least)

I tried to 'fix' this by calibrating the voltage readings to the know value of the LiPo charge cutoff voltage 4.2v. When your badge is connected to a charger and the battery is full, it should calibrate the voltage readings to improve offsets.

Please give it a try :)